MarkFzp/act-plus-plus #2: Missing packages to run?

ramkumarkoppu, 2024-01-03:
Followed the installation procedure to install listed pip packages in the condo environment but still get missing packages errors when I run `python3 imitate_episodes.py`

```
(aloha) Rams-MBP:act-plus-plus ramkumarkoppu$ python3 imitate_episodes.py
ROBOMIMIC WARNING(
    No private macro file found!
    It is recommended to use a private macro file
    To setup, run: python /Users/ramkumarkoppu/miniforge3/envs/aloha/lib/python3.8/site-packages/robomimic/scripts/setup_macros.py
)
Traceback (most recent call last):
  File "imitate_episodes.py", line 20, in <module>
    from policy import ACTPolicy, CNNMLPPolicy, DiffusionPolicy
  File "/Users/ramkumarkoppu/GIT/GitHub/Robotic_projs/act-plus-plus/policy.py", line 12, in <module>
    from robomimic.algo.diffusion_policy import replace_bn_with_gn, ConditionalUnet1D
ModuleNotFoundError: No module named 'robomimic.algo.diffusion_policy'
(aloha) Rams-MBP:act-plus-plus ramkumarkoppu$ 
```

SherwinDengxiong, 2024-01-04:
seems like there is no diffusion policy in the robomimic.algo

ramkumarkoppu, 2024-01-04:
How to fix this error?

SherwinDengxiong, 2024-01-04:
Currently, I skip the Diffusion part and waiting for the author response.

apirrone, 2024-01-04:
It seems that the diffusion policy in robomimic is available in a specific branch of the repo.

I was able to go further in running this with the following actions: 
- pip uninstall robomimic
- cd <somewhere> && git clone `git@github.com:ARISE-Initiative/robomimic.git`
- cd <somewhere>/robomimic && git checkout diffusion-policy-mg
- pip install -e .
- pip install diffusers

I also had to edit the `DATA_DIR` path in act-plus-plus/constants.py.

Then, it seems that the argument `--num_epochs` was replaced by `--num-steps`, so I changed it so that the command I run is : 

```bash
$ python3 imitate_episodes.py --task_name sim_transfer_cube_scripted --ckpt_dir trainings --policy_class ACT --kl_weight 10 --chunk_size 100 --hidden_dim 512 --batch_size 8 --dim_feedforward 3200  --lr 1e-5 --seed 0 --num_steps 2000 --onscreen_render
```

Now I get a new error, that I will reference in a new issue

WangKeAlchemist, 2024-01-04:
> How to fix this error?

Hi I tried to comment out line 12 and it is fine. Also you can install diffuser library as mentioned by @apirrone to avoid errors in following lines. But I still get other errors

WangKeAlchemist, 2024-01-04:
Also to load the demo dataset, you have to change `DATA_DIR` in `constants.py` and wandb settings at https://github.com/MarkFzp/act-plus-plus/blob/main/imitate_episodes.py#L148C41-L148C41

ramkumarkoppu, 2024-01-06:
> It seems that the diffusion policy in robomimic is available in a specific branch of the repo.
> 
> I was able to go further in running this with the following actions:
> 
> * `pip uninstall robomimic`
> * cd `somewhere` && `git clone git@github.com:ARISE-Initiative/robomimic.git`
> * cd `somewhere/robomimic` && `git checkout diffusion-policy-mg`
> * `pip install -e .`
> * `pip install diffusers`
> 
> I also had to edit the `DATA_DIR` path in `act-plus-plus/constants.py`
> 
> Then, it seems that the argument `--num_epochs` was replaced by `--num-steps`, so I changed it so that the command I run is :
> 
> ```shell
> $ python3 imitate_episodes.py --task_name sim_transfer_cube_scripted --ckpt_dir trainings --policy_class ACT --kl_weight 10 --chunk_size 100 --hidden_dim 512 --batch_size 8 --dim_feedforward 3200  --lr 1e-5 --seed 0 --num_steps 2000 --onscreen_render
> ```
> 
> Now I get a new error, that I will reference in a new issue [#4 (comment)](https://github.com/MarkFzp/act-plus-plus/issues/4#issue-2065393653)

How did you get checkpoints? `--ckpt_dir trainings`

apirrone, 2024-01-06:
Hi @ramkumarkoppu, for now I train from scratch, so the `trainings` dir is empty at start and checkpoints are saved here

mustang66ytz, 2024-01-10:
[image]
I encounter the wandb issue, how to solve this?

ramkumarkoppu, 2024-01-14:
got the same error

ramkumarkoppu, 2024-01-14:
> Hi @ramkumarkoppu, for now I train from scratch, so the `trainings` dir is empty at start and checkpoints are saved here

getting this 'wandb' error:
wandb: ERROR Error while calling W&B API: project not found (<Response [404]>)
Problem at: imitate_episodes.py 148 main
wandb: ERROR It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)
Traceback (most recent call last):
  File "imitate_episodes.py", line 666, in <module>
    main(vars(parser.parse_args()))
  File "imitate_episodes.py", line 148, in main
    wandb.init(project="mobile-aloha2", reinit=True, entity="mobile-aloha2", name=expr_name)
    
how did you get this working?

ramkumarkoppu, 2024-01-15:
> Also to load the demo dataset, you have to change `DATA_DIR` in `constants.py` and wandb settings at https://github.com/MarkFzp/act-plus-plus/blob/main/imitate_episodes.py#L148C41-L148C41

What's the example value to be set for DATA_DIR ?

tanxr00, 2024-01-17:
> Followed the installation procedure to install listed pip packages in the condo environment but still get missing packages errors when I run `python3 imitate_episodes.py`
>
> ```
> (aloha) [...]

已经解决，请使用这个分支https://github.com/ARISE-Initiative/robomimic/tree/diffusion-policy-mg

sainavaneet, 2024-04-04:
> ![image](https://private-user-images.githubusercontent.com/24708005/295489600-5ab00820-6ee0-4eec-93c7-133b53cf1fbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

use your own entity and project name