MarkFzp/act-plus-plus #3: TypeError: forward() got an unexpected keyword argument 'src_key_padding_mask'

luozhiping, 2024-01-04:
param error use imitate_episodes.py to train model.

```
TypeError: forward() got an unexpected keyword argument 'src_key_padding_mask'
TypeError: forward() got an unexpected keyword argument 'pos'

at detr_vae.py line 116:          encoder_output = self.encoder(encoder_input, pos=pos_embed, src_key_padding_mask=is_pad)

where transformer.py forward function param is     
def forward(self, src, mask, query_embed, pos_embed, latent_input=None, proprio_input=None, additional_pos_embed=None)

```
Are there any code updated haven't pushed?

luozhiping, 2024-01-04:
change input param to self.encoder(encoder_input, pos_embed=pos_embed, mask=is_pad) and then got this error:
: forward() missing 1 required positional argument: 'query_embed'

luozhiping, 2024-01-04:
[image]

is there any wrong here? actually it should be encoder = build_encoder(args)

apirrone, 2024-01-04:
Thanks @luozhiping 

Did just that and it seems to be training now !

in detr/models/detr_vae.py at line 285 change `build_transformer` to `build_encoder`

WangKeAlchemist, 2024-01-04:
> Thanks @luozhiping
> 
> Did just that and it seems to be training now !
> 
> in detr/models/detr_vae.py at line 285 change `build_transformer` to `build_encoder`

This works for me as well!

hszhoushen, 2024-11-06:
> This works for me as well!

> ![image](https://private-user-images.githubusercontent.com/4124006/294112454-e006fbf2-0f3c-4f24-b72f-2b3b2f93b7a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

This works for me as well!

Cryon233, 2024-11-09:
Thank you!This works for me as well!

Cryon233, 2024-11-09:
> Thanks @luozhiping
> 
> Did just that and it seems to be training now !
> 
> in detr/models/detr_vae.py at line 285 change to `build_transformer``build_encoder`

Thank you!This works for me as well!

Skylar-xty, 2024-11-12:
many thanks to you!

> ![image](https://private-user-images.githubusercontent.com/4124006/294112454-e006fbf2-0f3c-4f24-b72f-2b3b2f93b7a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]