MarkFzp/act-plus-plus #4: Error loading episode_XX.hdf5 in __getitem__

apirrone, 2024-01-04:
Hi!

After performing some changes described here : https://github.com/MarkFzp/act-plus-plus/issues/2#issuecomment-1876860913

When running the `imitate_episodes.py` script, I get the following error:

[image]

I displayed the exception text in the code and got :
`"Unable to synchronously open object (object 'left_wrist' doesn't exist)"`

Any idea of what is going on ? 

Thanks !

apirrone, 2024-01-04:
Maybe I misunderstood, but I just downloaded the data here https://drive.google.com/drive/folders/1gPR03v05S1xiInoVJn7G7VJ9pDCnxq9O 

Should I run `record_sim_episodes.py` first ? I thought it would just regenerate the .hdf5 files

apirrone, 2024-01-04:
Indeed, after `running record_sim_episodes.py` I get a different error. I will create a new issue for that, closing this one  :)

closed 2024-01-04 by apirrone

dva1213, 2024-01-30:
Hello, I also encountered this problem when running training, but I see that you have no steps to solve this problem, may I ask how you solved it
[image]

zxfever, 2024-04-08:
> Hello, I also encountered this problem when running training, but I see that you have no steps to solve this problem, may I ask how you solved it ![779ad549cf9034ddc79eb59e7fae763](https://private-user-images.githubusercontent.com/142143304/300716821-73257b5f-7e1e-4f58-ab9d-9e3886ce7298.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NDIxNjcsIm5iZiI6MTcxMjU0MTg2NywicGF0aCI6Ii8xNDIxNDMzMDQvMzAwNzE2ODIxLTczMjU3YjVmLTdlMWUtNGY1OC1hYjlkLTllMzg4NmNlNzI5OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOFQwMjA0MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNmUwYmIyZmVjZDc0MTIwYmE4ODQwNDQ4YzkzMjhkMDZjMTdiZWFjODc3Nzg4OTIwNTU0ZWFmMGQwMjZiOThkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.rD90Iazw2-qku_xM5ICIoLq9FYKgMFNXnPrec2Lfz3c)

Hello, the same error, have you solved it?