MarkFzp/act-plus-plus #11: Performance in simulation environment is not very good

Boxjod, 2024-01-08:
Follow the environment configuration steps provided by your code repository to complete 50 simulations of the two tasks. The success rate of the first task is 10/50, and the success rate of the second task is 1/50. There is also a mirror task in the code. The success rate is 15/50.

But for the task of passing the wooden block, does it count as a success if it touches two robotic arms? Because I visualized and videoed each data, none of the tasks of passing the wooden blocks were truly successful.

I want to know if this is the case for everyone? GPU computing is not used in my environment.

[image]

[image]

WangKeAlchemist, 2024-01-08:
I have the same issue, i don not think that is the GPU issue for generating demos

WangKeAlchemist, 2024-01-08:
> Follow the environment configuration steps provided by your code repository to complete 50 simulations of the two tasks. The success rate of the first task is 10/50, and the success rate of the second task is 1/50. There is also a mirror task in the code. The success rate is 15/50.
> 
> But for the task of passing the wooden block, does it count as a success if it touches two robotic arms? Because I visualized and videoed each data, none of the tasks of passing the wooden blocks were truly successful.
> 
> I want to know if this is the case for everyone? GPU computing is not used in my environment.
> 
> ![result2](https://private-user-images.githubusercontent.com/21015502/294829400-5af37545-b95d-44ad-8f1c-d9d9ceb09bb0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MDk0MjEsIm5iZiI6MTcwNDcwOTEyMSwicGF0aCI6Ii8yMTAxNTUwMi8yOTQ4Mjk0MDAtNWFmMzc1NDUtYjk1ZC00NGFkLThmMWMtZDlkOWNlYjA5YmIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDEwMTg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiNjljOWM4NGY1MjM4ZTkxM2FkNmMyMzcxZGVhNTFkZTU1ZTc5MGUwYmUyMmE5NTc5ZTk5MjNlZmQwMmYxOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.48HQhs-ncWshaUd28UAG4l7CmChdbQ8lCj_Af-uYnw8)
> 
> ![result](https://private-user-images.githubusercontent.com/21015502/294829295-de03a66f-3da6-4e93-9ffd-77e7e89a4a66.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MDk0MjEsIm5iZiI6MTcwNDcwOTEyMSwicGF0aCI6Ii8yMTAxNTUwMi8yOTQ4MjkyOTUtZGUwM2E2NmYtM2RhNi00ZTkzLTlmZmQtNzdlN2U4OWE0YTY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDEwMTg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1NjgyZmNhMWU2ZjhhNTU3MWIxMGVmMzIxZDFiMTRhZTgwMTZkZWI3NGEwYzVjOTM0ZTk2YTM0ZjgzYThkZjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TsIjdEb61pOd6_jVyluX8Zh7Z0OkTK_ZeFIc3xbcJU8)

Yeah i think if the cube touches the left arm gripper it will be considered as successful

tonyzhaozh, 2024-01-11:
I am suspecting the scripted data generation has bug here. The success should be 50/50. Zipeng and I will look into this. 

It is worth noting in the meantime that the sim component of this codebase is NOT used actively. It is essentially the same as the original ALOHA/ACT repo: https://github.com/tonyzhaozh/act.

tonyzhaozh, 2024-01-11:
I am closing this issue as there are currently multiple issues pointing to the same dataset generation problem. Please check out https://github.com/MarkFzp/act-plus-plus/issues/12 where I will update progress on our end in fixing this.

Thanks,
Tony

closed 2024-01-11 by tonyzhaozh