MarkFzp/act-plus-plus #14: ERROR Error while calling W&B API: project not found (<Response [404]>)

super2013sky, 2024-01-09:
wandb: W&B API key is configured. Use `wandb login --relogin` to force relogin
wandb: ERROR Error while calling W&B API: project not found (<Response [404]>)
Problem at: imitate_episodes.py 148 main
wandb: ERROR It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)
Traceback (most recent call last):
  File "imitate_episodes.py", line 666, in <module>
    main(vars(parser.parse_args()))
  File "imitate_episodes.py", line 148, in main
    wandb.init(project="mobile-aloha2", reinit=True, entity="mobile-aloha2", name=expr_name)
  File "/home/hf/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 1189, in init
    raise e
  File "/home/hf/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 1170, in init
    run = wi.init()
  File "/home/hf/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 781, in init
    raise error
wandb.errors.CommError: It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)

************************************************************************************
The above is an error message. How can I obtain permission for mobile aloha2 and solve this problem

WangKeAlchemist, 2024-01-09:
Hi you have to change the entity to your own wandb entity at https://github.com/MarkFzp/act-plus-plus/blob/main/imitate_episodes.py#L148

super2013sky, 2024-01-10:
> Hi you have to change the entity to your own wandb entity at https://github.com/MarkFzp/act-plus-plus/blob/main/imitate_episodes.py#L148

Okay, I understand now, it's been resolved, thank you.

closed 2024-01-10 by super2013sky