MarkFzp/act-plus-plus #15: Simulation issues

deavn2236, 2024-01-09:
The data generated by "record_sim_episodes. py" mostly fails. In the project, there is "sim_env. py" that can enable the simulation environment. However, even after configuring Interbotix, the following errors still exist. How can I build data through the simulation environment and test the program without hardware

Python sim_ Env.py
Timeout exceeded while waiting for service/master_ Left/set_ Operating_ Modes
The robot 'master_ Left 'is not discoverable Did you enter the correct robot_ Name parameter? Is the xs_ SDK node running? Quitting

sjz960913, 2024-01-10:
I have got the same problem,did you solve that?

tonyzhaozh, 2024-01-11:
Something is off with the current scripted data generation, and we are currently fixing it. You should not try to run python sim_env.py as that is for a legacy test in the original ALOHA codebase (it uses real ALOHA leader arms to control sim ALOHA follower arms, it won't work unless you have two real leader arms connected to your computer.) 

Please follow this thread (https://github.com/MarkFzp/act-plus-plus/issues/12) in which I will update our progress on solving this issue. Right now there are multiple issues pointing to the same dataset problem.

Thanks,
Tony

closed 2024-01-11 by tonyzhaozh

lijingshanxi, 2024-03-30:
> Something is off with the current scripted data generation, and we are currently fixing it. You should not try to run python sim_env.py as that is for a legacy test in the original ALOHA codebase [...]

Hi @tonyzhaozh ,

Thank you for your effort. I would like to ask how to use the keyboard to operate the sim arm. Is there any similar code that I can refer to;