MarkFzp/act-plus-plus #16: Error: Visualize_episodes

Doan-IT, 2024-01-10:
Hello. 
- Following the readme.txt instructions, I tried running visualize the episode:
python visualize_episodes.py --dataset_dir dataset/aloha_mobile_wipe_wine --episode_idx 10

- data save dir: download "aloha_mobile_wipe_wine" from google drive. (episode_0.hdf5 - to episode_49.hdf5)

- result: numpy.exceptions.AxisError:
all_cam_videos = np.concatenate(all_cam_videos, axis=2) # width dimension
numpy.exceptions.AxisError: axis 2 is out of bounds for array of dimension 2

Please support. Thanks a lot!

lwcode, 2024-01-10:
meet the same problem

tonyzhaozh, 2024-01-11:
Sorry for the confusion: the visualize script here is used for the legacy sim envs. Please use this https://github.com/MarkFzp/mobile-aloha/blob/main/aloha_scripts/visualize_episodes.py instead.

closed 2024-01-11 by tonyzhaozh