MarkFzp/act-plus-plus #19: Paper explain

Doan-IT, 2024-01-11:
Thank you for sharing such a great project.
Refer to the paper "https://mobile-aloha.github.io/resources/mobile-aloha.pdf" I have some questions, hoping for your help:
1. What is the definition of "Co-train"? Source corresponds to "co-train"
2. "VINN + Chunking" is "CNNMLP" in the source code, right?

Thanks a lot.

kuaile258890, 2024-01-16:
2. In the source code, VINN + Chunking should not be confused with CNNMLP. They represent different baselines. For more details, refer to the original ALOHA paper (not to be mistaken with mobile-ALOHA).

kuaile258890, 2024-01-16:
In the original ALOHA paper, BC-ConvMLP (also known as CNNMLP) is identified as the simplest yet most widely used baseline [69, 26]. It processes the current image observations with a convolutional network. The output features of this network are then concatenated with the joint positions to predict the action."