MarkFzp/act-plus-plus #20: use the offical dataset, the hdf5 files is error

VDuZane, 2024-01-12:
Hi, thanks for your brilliant work and open-source code!
I tried your dataset files to generate mp4 file, or to train the model. 
[https://drive.google.com/drive/folders/1vhKa4tNPfMkeK7SW86vKorb39epTYW39] —— aloha_mobile_chair_truncated
program running error, error code is :
Traceback (most recent call last):
  File "visualize_episodes.py", line 160, in <module>
    main(vars(parser.parse_args()))
  File "visualize_episodes.py", line 44, in main
    save_videos(image_dict  , DT, video_path=os.path.join(dataset_dir, dataset_name + '_video.mp4'))
  File "visualize_episodes.py", line 78, in save_videos
    all_cam_videos = np.concatenate(all_cam_videos, axis=2) # width dimension
  File "<__array_function__ internals>", line 200, in concatenate
numpy.AxisError: axis 2 is out of bounds for array of dimension 2
The real environment file data format is not same as simulation generated data files especially the camera data.
How to solve this problem?
[image]
[image]

tanxr00, 2024-01-17:
采集数据中，图像压缩模式关掉

huiwenzhang, 2024-01-23:
You shold use the visualize_episodes.py script hosted in https://github.com/MarkFzp/mobile-aloha/tree/main/aloha_scripts if you want to visualize real-world collected data from the author. The visualize_episodes.py in this repo only used for visualize simulated task like sim_tansfer_cube and sim_insertion_scripted.

edcand3ball, 2024-06-15:
Hello, may I ask how to modify the data format of my robotic arm to training format, and can the data of a six axis robotic arm be trained