MarkFzp/act-plus-plus #21: DIffusion Policy parameters

yxKryptonite, 2024-01-12:
Hi authors,

I used your 50 demo episodes to train ACT and it worked very well, achieving success rate up to 90% on the cube-transferring task. However, after I changed the algorithm to Diffusion Policy, it turned out that diffusion policy had very low success. I tried multiple hyperparameter settings in your `commands.txt` but they all couldn't work. The results are below:

[image]

(The green is ACT, while others are Diffusion Policy with different hyperparam sets)

**So I wonder why that happens and could you share the best-working diffusion policy parameters? Thank you very much!**

yxKryptonite, 2024-01-20:
Hi authors, thanks for your work! But I tried multiple sets of diffusion policy parameters but they all couldn't work? Could you please kindly guide me how to train diffusion policy on the cube-transferring task or provide any trained checkpoints? Thank you very much!

Wallong, 2024-04-02:
@yxKryptonite, the same question. Have you tried the command like this? 
`conda activate mobile
export MUJOCO_GL=egl
cd /home/tonyzhao/Research/act-plus-plus
CUDA_VISIBLE_DEVICES=0 python3 imitate_episodes.py \
--task_name sim_transfer_cube_scripted \
--ckpt_dir /scr/tonyzhao/train_logs/cube_scripted_diffusion_sweep_4_regressionTest \
--policy_class Diffusion --chunk_size 32 \
--batch_size 32 --lr 1e-4 --seed 0 \
--num_steps 200000 --eval_every 6000 --validate_every 6000 --save_every 6000`

barsm42, 2024-07-08:
Hello, 

Same question here. I trained Ziploc Slide ( we created our dataset but same task), and ACT worked well for the task. Then we tried Diffusion Policy class. The validation losses of Diffusion Policy were way much better than ACT for the training. But during the inference, Diffusion Policy did not work. Robots could not even start for the task. I will share the error later to ask. It was an error about "self.ema.averaged_model".

`
class DiffusionPolicy(nn.Module):
    def __init__(self, args_override):
        super().__init__()

        self.camera_names = args_override['camera_names']
        self.observation_horizon = args_override['observation_horizon'] ### TODO TODO TODO DO THIS
        self.action_horizon = args_override['action_horizon'] # apply chunk size
        self.prediction_horizon = args_override['prediction_horizon'] # chunk size
        self.num_inference_timesteps = args_override['num_inference_timesteps']
        self.ema_power = args_override['ema_power']
        self.lr = args_override['lr']
        self.weight_decay = 0

        self.num_kp = 32
        self.feature_dimension = 64
        self.ac_dim = args_override['action_dim'] # 14 + 2
        self.obs_dim = self.feature_dimension * len(self.camera_names) + 14 # camera features and proprio

        backbones = []
        pools = []
        linears = []
        for _ in self.camera_names:
            backbones.append(ResNet18Conv(**{'input_channel': 3, 'pretrained': False, 'input_coord_conv': False}))
            pools.append(SpatialSoftmax(**{'input_shape': [512, 15, 20], 'num_kp': self.num_kp, 'temperature': 1.0, 'learnable_temperature': False, 'noise_std': 0.0}))
            linears.append(torch.nn.Linear(int(np.prod([self.num_kp, 2])), self.feature_dimension))
        backbones = nn.ModuleList(backbones)
        pools = nn.ModuleList(pools)
        linears = nn.ModuleList(linears)
        
        backbones = replace_bn_with_gn(backbones) # TODO

        noise_pred_net = ConditionalUnet1D(
            input_dim=self.ac_dim,
            global_cond_dim=self.obs_dim*self.observation_horizon
        )

        nets = nn.ModuleDict({
            'policy': nn.ModuleDict({
                'backbones': backbones,
                'pools': pools,
                'linears': linears,
                'noise_pred_net': noise_pred_net
            })
        })

        nets = nets.float().cuda()
        ENABLE_EMA = True
        if ENABLE_EMA:
            ema = EMAModel(parameters=nets, power=self.ema_power)#power=self.ema_power
        else:
            ema = None
        self.nets = nets
        self.ema = ema

        # setup noise scheduler
        self.noise_scheduler = DDIMScheduler(
            num_train_timesteps=50,
            beta_schedule='squaredcos_cap_v2',
            clip_sample=True,
            set_alpha_to_one=True,
            steps_offset=0,
            prediction_type='epsilon'
        )

        n_parameters = sum(p.numel() for p in self.parameters())
[... 88 lines cut ...]
                    model_output=noise_pred,
                    timestep=k,
                    sample=naction
                ).prev_sample

            return naction

    def serialize(self):
        return {
            "nets": self.nets.state_dict(),
            "ema": self.ema.averaged_model.state_dict() if self.ema is not None else None,
        }

    def deserialize(self, model_dict):
        status = self.nets.load_state_dict(model_dict["nets"])
        print('Loaded model')
        if model_dict.get("ema", None) is not None:
            print('Loaded EMA')
            status_ema = self.ema.averaged_model.load_state_dict(model_dict["ema"])
            status = [status, status_ema]
        return status

`

Training command: python3 imitate_episodes.py --task_name aloha_slide_exp1 --ckpt_dir C:/Users/aa/Desktop/act-main/ckpt --policy_class DiffusionPolicy --kl_weight 10 --chunk_size 100 --hidden_dim 512 --batch_size 4 --dim_feedforward 3200 --num_epochs 200  --lr 1e-4 --seed 0

woltium, 2024-07-10:
@barsm42  same question, Have you solved this problem?

barsm42, 2024-07-11:
> @barsm42 same question, Have you solved this problem?

We are trying to solve. I trained two policies with same parameters. The only difference is "ENABLE_EMA = True" or "ENABLE_EMA = False" line. We will evaluate, and check if it gives EMAModel error during the inference.

lanlankilkil, 2024-08-14:
Hello, I am also replicating aloha diffusion, saying that a parameter is required, then I change the model in ema=EMAMODEL (model=nets, power=self.ema.power) to parameters, After the previous error was reported, but later it was reported that EMAMODEL object no average_model, can you add a contact information for help

lanlankilkil, 2024-08-14:
@yxKryptonite Hello, I am also replicating aloha diffusion, saying that a parameter is required, then I change the model in ema=EMAMODEL (model=nets, power=self.ema.power) to parameters, After the previous error was reported, but later it was reported that EMAMODEL object no average_model, can you add a contact information for help

lanlankilkil, 2024-08-14:
> Hello,
>
> Same question here. I trained Ziploc Slide ( we created our dataset but same task), and ACT worked well for the task. Then we tried Diffusion Policy class. The validation losses of Diffusion [...]
@barsm42 Hello, I am also replicating aloha diffusion, saying that a parameter is required, then I change the model in ema=EMAMODEL (model=nets, power=self.ema.power) to parameters, After the previous error was reported, but later it was reported that EMAMODEL object no average_model, can you add a contact information for help

brisyramshere, 2024-09-05:
> @yxKryptonite Hello, I am also replicating aloha diffusion, saying that a parameter is required, then I change the model in ema=EMAMODEL (model=nets, power=self.ema.power) to parameters, After the [...]

have you addressed this problem?

huanxulin1, 2024-11-01:
There is a lerobot repository in Huggingface that can be used to train diffusion policies in the Aloha environment. You can take a look

AbdElrahmanMostafaRifaat1432, 2024-12-25:
have anyone find a solution to the problem yet

return-sleep, 2025-02-06:
same question,  diffusion policy has bad performance.