MarkFzp/act-plus-plus #22: Issue about training the model

ddliu365, 2024-01-13:
Hello, folks,

I met an issue while training the model. 

It seems caused by the permission to the project.

```
(aloha) jack@jack:~/repos/act-plus-plus$ python3 imitate_episodes.py --task_name sim_transfer_cube_scripted --ckpt_dir ./ --policy_class ACT --kl_weight 10 --chunk_size 100 --hidden_dim 512 --batch_size 8 --dim_feedforward 3200  --lr 1e-5 --seed 0 --num_steps 10
wandb: (1) Create a W&B account
wandb: (2) Use an existing W&B account
wandb: (3) Don't visualize my results
wandb: Enter your choice: 1
wandb: You chose 'Create a W&B account'
wandb: Create an account here: https://wandb.ai/authorize?signup=true
wandb: Paste an API key from your profile and hit enter, or press ctrl+c to quit: 
wandb: Appending key for api.wandb.ai to your netrc file: /home/jack/.netrc
wandb: ERROR Error while calling W&B API: project not found (<Response [404]>)
Problem at: imitate_episodes.py 148 main
wandb: ERROR It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)
Traceback (most recent call last):
  File "imitate_episodes.py", line 666, in <module>
    main(vars(parser.parse_args()))
  File "imitate_episodes.py", line 148, in main
    wandb.init(project="mobile-aloha2", reinit=True, entity="mobile-aloha2", name=expr_name)
  File "/home/jack/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 1195, in init
    raise e
  File "/home/jack/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 1176, in init
    run = wi.init()
  File "/home/jack/anaconda3/envs/aloha/lib/python3.8/site-packages/wandb/sdk/wandb_init.py", line 785, in init
    raise error
wandb.errors.CommError: It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)
```

ramkumarkoppu, 2024-01-14:
same issue

junhui1997, 2024-01-15:
create a wandb account and modify it accordingly or just disable wandb 
`os.environ["WANDB_DISABLED"] = "true"`

tanxr00, 2024-01-17:
> same issue

hi, how to infer a scene?

tanxr00, 2024-01-17:
> same issue

hi，are U trained custom real data？

destroy314, 2024-04-25:
https://github.com/MarkFzp/act-plus-plus/issues/2#issuecomment-2076723911
https://github.com/MarkFzp/act-plus-plus/issues/14#issuecomment-1882681746

fderxs, 2024-05-29:
In imitate_episodes.py, line 148:
`if not is_eval:
        wandb.init(project="mobile-aloha2", reinit=True, entity="mobile-aloha2", name=expr_name)
        wandb.config.update(config)`
Change the **entity** to your own wandb username or team name, like that:
`if not is_eval:
        wandb.init(project="mobile-aloha2", reinit=True, entity="<your wandb name>", name=expr_name)
        wandb.config.update(config)`