MarkFzp/act-plus-plus #23: eval success rate 0.5

lwcode, 2024-01-15:
[image]

task is sim_transfer_cube_scripted, 50 episodes, trainning parameters:
`python3 imitate_episodes.py --task_name sim_transfer_cube_scripted \
--ckpt_dir ./models --policy_class ACT --kl_weight 10 --chunk_size 100 \
--hidden_dim 512 --batch_size 8 --dim_feedforward 3200 --num_steps 2000 \
--lr 1e-5 --seed 0 --eval`
Is the eval result normal?

lwcode, 2024-01-15:
My fault, need modify '--num_steps' to greater number.

closed 2024-01-15 by lwcode

lunalulu, 2024-01-19:
> My fault, need modify '--num_steps' to greater number.

Why does the success rate decrease when I increase the number of training sessions?

lwcode, 2024-01-19:
> > My fault, need modify '--num_steps' to greater number.
> 
> Why does the success rate decrease when I increase the number of training sessions?

success rate seems to converge to fluctuate around a certain value for large `num_step`, for example 20000