MarkFzp/act-plus-plus #28: co_training data

a510721, 2024-01-18:
I have a question about co-training 
looking at the constants.py, you can see the following dataset_dir aloha_compressed_dataset
is the aloha_compressed_dataset the same as the aloha_static_contraining _dataset for co_trainning?

it' too slow to train whole dataset(aloha_mobile_elevator_truncated+aloha_static_contraining _dataset)
how is it a good idea to organize the dataset?

"aloha_mobile_elevator_2 and aloha_mobile_elevator_button" there is no data. can you share the missing data?

    'aloha_mobile_elevator_truncated_cotrain':{
        'dataset_dir': [
            DATA_DIR + '/aloha_mobile_elevator_truncated',
            DATA_DIR + '/aloha_mobile_elevator_2',
            DATA_DIR + '/aloha_mobile_elevator_button',
            DATA_DIR + '/aloha_compressed_dataset',
        ], # only the first dataset_dir is used for val
        'stats_dir': [
            DATA_DIR + '/aloha_mobile_elevator_truncated',
            DATA_DIR + '/aloha_mobile_elevator_2',
        ],
        'sample_weights': [3, 3, 2, 1],
        'train_ratio': 0.99, # ratio of train data from the first dataset_dir
        'episode_len': 2250,
        'camera_names': ['cam_high', 'cam_left_wrist', 'cam_right_wrist']
    },

lunalulu, 2024-01-19:
Why don't I see the code above about co training?

huiwenzhang, 2024-01-25:
> Why don't I see the code above about co training?

The code above is used for real-world dataset training, and the source code is hosted in https://github.com/MarkFzp/mobile-aloha/blob/main/aloha_scripts/constants.py. This repo only includes codes for training in simulation environment.