MarkFzp/act-plus-plus #31: Train ACT with aloha_mobile_wipe_wine dataset. Error: No module named 'aloha_scripts'

super2013sky, 2024-01-25:
[image]
[image]

can not find this file: aloha_scripts.constants

Thank you

Magician-1024, 2024-01-25:
https://github.com/MarkFzp/mobile-aloha
aloha_scripts is in this repo, but I dont know exactly how to use it

tanxr00, 2024-02-04:
> ![QQ20240125-102746](https://private-user-images.githubusercontent.com/3701836/299519537-1d49881c-06c0-4bc5-bb06-284e2e997eb1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNDA3NjEsIm5iZiI6MTcwNzA0MDQ2MSwicGF0aCI6Ii8zNzAxODM2LzI5OTUxOTUzNy0xZDQ5ODgxYy0wNmMwLTRiYzUtYmIwNi0yODRlMmU5OTdlYjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDRUMDk1NDIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODI1MzgyZWQ1MWUwYTU5ZDU1MThmOTRmMzNhZDdiYzVhMWUwOGU5Nzg3YTlhZmEyYThlODg5MzJiYTgzYjgxYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pA8L4rlp0g_qNePD0G1eoeEjhT7Xc2mHZiF-9sh52EI) ![QQ20240125-103958](https://private-user-images.githubusercontent.com/3701836/299520287-5cabc53b-9004-45c4-84a2-10eceee97bc2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNDA3NjEsIm5iZiI6MTcwNzA0MDQ2MSwicGF0aCI6Ii8zNzAxODM2LzI5OTUyMDI4Ny01Y2FiYzUzYi05MDA0LTQ1YzQtODRhMi0xMGVjZWVlOTdiYzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDRUMDk1NDIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTgwZjZiNDkwYWZlMTgwYTJhZGExOTAwMjI4OTAwMDk4YjYxNDZmOTZlMGMyZGQ0MTQyZjU2YWUzMTkwM2I4ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.JgMOxp2PJb_21CDdb_5Zf586sfstK6qTG1dJrf-eIys)
> 
> can not find this file: aloha_scripts.constants
> 
> Thank you

---
终端运行export PYTHONPATH=$PYTHONPATH:"./"，然后执行python训练代码
# 或者 python源码中添加
import sys  # 添加python环境
sys.path.append("./")