MarkFzp/act-plus-plus #45: Error loading /home/robot/refer/act-plus-plus/data/sim_transfer_cube_scripted/episode_10.hdf5 in __getitem__

dbdxnuliba, 2024-03-27:
Error loading /home/robot/refer/act-plus-plus/data/sim_transfer_cube_scripted/episode_10.hdf5 in __getitem__

theworld25, 2024-04-07:
Have you solved this problem?

theworld25, 2024-04-07:
这个好像是数据集下载时包含了隐藏的.episode_x.hdf5文件，我是把这个删除就ok了

zxfever, 2024-04-08:
The same error, have you solved it?

theworld25, 2024-04-08:
Sorry, my previous description was wrong, but I also encountered this problem and haven't solved it yet.

zxfever, 2024-04-08:
> The same error, have you solved it?
I used the  `record_episodes.py` file to generate 50 simulation data, and loaded the generated data is ok.

theworld25, 2024-04-15:
是因为作者提供的数据集都是一个相机的数据，把constants.py中只保留top，其他的删除就OK了

theworld25, 2024-04-15:
The dataset provided by the author contains only one camera, and it is enough to delete the other two in the constants.py except for the top.

sainavaneet, 2024-06-12:
> The dataset provided by the author contains only one camera, and it is enough to delete the other two in the constants.py except for the top.

this worked thank you