MarkFzp/act-plus-plus #48: Lack of image normalization in DiffusionPolicy

hesic73, 2024-04-13:
In `policy.py`, image normalization is applied in both ACTPolicy and CNNMLPPolicy as follows:

```python
        normalize = transforms.Normalize(mean=[0.485, 0.456, 0.406],
                                         std=[0.229, 0.224, 0.225])
        image = normalize(image)
```

However, this normalization step is missing in DiffusionPolicy. Is this omission intentional? Could you clarify the reasoning behind this difference?

woltium, 2024-07-10:
> In `policy.py`, image normalization is applied in both ACTPolicy and CNNMLPPolicy as follows:
>
> ```python
>         normalize = transforms.Normalize(mean=[0.485, 0.456, [...]

@hesic73  same  question,and  whether the normalization is added or not, the final eval performance of Diffusion policy is  poor,arms keeps shaking and can't complete the task. Have you solved this problem?

QueirosJustin, 2024-07-13:
@woltium Hello, How did Diffusion Policy even run for you? Even after following all the instructions I get an error until I turn off         ENABLE_EMA = True in policy.py to false. Once I do this though, I receive  " raise _control.PhysicsError(message)
dm_control.rl.control.PhysicsError: Physics state is invalid. Warning(s) raised: mjWARN_BADQACC" everytime during the first evaluation. For ACT everything is working fine for me. I would appreciate an answer a lot.

QueirosJustin, 2024-07-13:
@woltium I used this command: python3 imitate_episodes.py --task_name sim_transfer_cube_scripted --ckpt_dir /home/Desktop/learning --policy_class Diffusion --chunk_size 32 --batch_size 32 --lr 1e-5 --seed 0 --num_steps 100000 --eval_every 2000 --validate_every 2000 --save_every 2000

sqdy2669, 2025-03-23:
> > In `policy.py`, image normalization is applied in both ACTPolicy and CNNMLPPolicy as follows:在`policy.py`中，图像规范化应用于ACTPolicy和CNNMLPPolicy，如下所示：
> >         normalize = transforms.Normalize(mean=[0.485, 0.456, 0.406],
> >                                          std=[0.229, 0.224, 0.225])
> >         image = normalize(image)
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > However, this normalization step is missing in DiffusionPolicy. Is this omission intentional? Could you clarify the reasoning behind this difference?然而，这个标准化步骤在DiffusionPolicy中缺失。这是故意的吗？你能解释一下这种差异背后的原因吗？
> 
> [@hesic73](https://github.com/hesic73) same question,and whether the normalization is added or not, the final eval performance of Diffusion policy is poor,arms keeps shaking and can't complete the task. Have you solved this problem?[@hesic73](https://github.com/hesic73)同样的问题，不管有没有加规范化，扩散策略的最终评估性能都很差，手臂一直在抖，无法完成任务。你解决这个问题了吗？
me too even in the diffusion-policy, Have you solved this problem?