MarkFzp/act-plus-plus #60: ModuleNotFoundError: No module named 'robomimic'

HJX-exoskeleton, 2024-10-19:
error:
Traceback (most recent call last):
  File "/home/song/hjx/aloha/mobile-aloha_multi-sensor/act-plus-plus/imitate_episodes.py", line 20, in <module>
    from policy import ACTPolicy, CNNMLPPolicy, DiffusionPolicy
  File "/home/song/hjx/aloha/mobile-aloha_multi-sensor/act-plus-plus/policy.py", line 11, in <module>
    from robomimic.models.base_nets import ResNet18Conv, SpatialSoftmax
ModuleNotFoundError: No module named 'robomimic'

TangGuohh, 2025-01-06:
look this :
also need to install https://github.com/ARISE-Initiative/robomimic/tree/r2d2 (note the r2d2 branch) for Diffusion Policy by pip install -e .

you need to git clone https://github.com/ARISE-Initiative/robomimic/tree/r2d2
and cd to  a correct path to run pip install -e .

TangGuohh, 2025-01-06:
small black son